sparkfun/SparkFun_SX1509_Arduino_Library #2: No need to wait for data after Wire.requestFrom()

Koepel, 2016-08-20:
The timeout or the while-loop to wait for something is not needed after Wire.requestFrom().

When the Wire.requestFrom() function returns, the I2C transaction has completely finished and the received data is in the receive buffer in the Wire library. The Wire.read() only reads data from that buffer.

When there was a I2C bus error or I2C collision, there would be less bytes received, and that while-loop would halt the sketch.

fluffy, 2019-08-05:
I'm running the 2.0.1 version which seems to be what Arduino IDE downloads as latest. I have a few other I2C things on the same bug and my sketch (running on a SAMD51)  locks up in this loop within a few minutes of starting. Be great to see a fix make it into a release.

TheNitek, 2020-07-07:
Any chance to get this released after 4 years?