sparkfun/SparkFun_SX1509_Arduino_Library #4: Servo support?

waseltzer, 2018-07-12:
Is it possible to add support for servos to the SX1509 library?
Seems that the PWM functionality could do this.
Need 50 Hz pulses between 1.0 and 2.0 milliseconds.

As implemented by: https://www.arduino.cc/en/reference/servo

viorels, 2020-02-24:
This is kind of late, but while looking for the answer to the same question I bumped into this page. Looking at the datasheet I see that the internal clock is 2MHz and divider is 3 bits so you can't generate less then a couple of kHz with this device, far from the required 50Hz. As I understand, there is an option to use an external clock (maybe 50Hz from an arduino pin?) and make it work, but I haven't tried it, it's only theoretical and not very elegant either.

raphael-bmec-co, 2024-08-06:
This is even later 😆 but there is the possibility of doing this using the blink functionality.