sparkfun/SparkFun_SX1509_Arduino_Library #14: Hardware Reset -> reset(1) might be broken

ajowsey, 2022-02-10:
Within init(), reset(1) is called if a hardware reset is provided:
}
   if (hardware) {
      // Check if bit 2 of REG_MISC is set
      // if so nReset will not issue a POR, we'll need to clear that bit first
      byte regMisc = readByte(REG_MISC);
      if (regMisc & (1 << 2)) {
         regMisc &= ~(1 << 2);
         writeByte(REG_MISC, regMisc);
      }
      // Reset the SX1509, the pin is active low
      **pinMode(pinReset, OUTPUT);     // set reset pin as output
      digitalWrite(pinReset, LOW);   // pull reset pin low
      delay(1);                      // Wait for the pin to settle
      digitalWrite(pinReset, HIGH);  // pull reset pin back high**
   }

The bolded lines of code above assume your Arduino will reset the io expander using the gpio assigned to pinReset. Problem is that this library actually calls SX1509::digitalWrite() instead of the Arduino digitalWrite. So the hardware reset never occurs

nseidle, 2022-02-10:
Thanks for reporting and I agree. I don't have a chance to fix it right now so if you do a PR, I would welcome it.

ajowsey, 2022-02-22:
Fix coming in this sprint hopefully! will share :)

nseidle, 2022-04-04:
Should be fixed in v3.0.4. Please check. Thanks for the PR!

closed 2022-04-04 by nseidle