sparkfun/SparkFun_SX1509_Arduino_Library #19: Compiler complains about SX1509::readWord(uint8_t registerAddress, uint16_t *value)

PKeller34, 2023-05-11:
I'm using Arduino 2.1.0 and version 3.0.4 of the SparkFunSX1509 library.

When compiling using #include <SparkFunSX1509.h> the compiler complains about an out of bounds array reference:

.....SX1509_IO_Expander\src\SparkFunSX1509.cpp:790:26: warning: array subscript 1 is outside array bounds of 'uint16_t [1]' {aka 'short unsigned int [1]'} [-Warray-bounds]
  790 |                 value[1] = dest[0];
      |                 ~~~~~~~~~^~~~~~~~~

There are a cascade of other complaints that are all triggered by the same basic issue: referencing a u_int16 as an array of u_int8[2].  The compiler understands what is intended, but complains because now it's stricter about types.

Replacing the code:

		value[0] = dest[1];
		value[1] = dest[0];
with:

               value[0] = (dest[1] << 8) + dest[0];

does the deed in a way that directly reflects the intent and keeps the compiler happy.

So the updated method looks like this (old code commented out):

bool SX1509::readWord(uint8_t registerAddress, uint16_t *value)
{
	uint8_t dest[2];
	if (readBytes(registerAddress, dest, 2))
	{
		// value[0] = dest[1];
		// value[1] = dest[0];                                    // "value[1]" is actually not defined, except in classic C
               value[0] = (dest[1] << 8) + dest[0];              // Type correct version
		return true;
	}
	return false;
}

Since I'm not yet experienced in git-ism's I am reluctant to just submit a pull request and make this change myself so I guess I'm punting this to the experts at SparkFun:-)

PaulZC, 2025-04-28:
Resolved in v3.0.6

Best,
Paul

PKeller34, 2025-04-29:
I had to invoke some Arduino IDE-Foo to get the Library Manager to see the update, but once that was resolved I could compile again, hack free!

Thank you!

Perry