sparkfun/SparkFun_SX1509_Arduino_Library #23: Multiple typecasting issues in this library.

yashmulgaonkar, 2023-06-22:


yashmulgaonkar, 2023-06-22:
Looks like this was caused by what looks like a careless "replace all" operation in this commit:
https://github.com/sparkfun/SparkFun_SX1509_Arduino_Library/commit/3ba49b6a5f39bb3d40db114dd3a54b010b804e7d

nseidle, 2023-06-22:
You're probably right. Maybe [you can fix it](https://youtu.be/velJRGZRCY0?t=149). PRs are always welcome.

gb-123-git, 2023-10-14:
@yashmulgaonkar 
I went through the earlier commits. I don't think this has been caused by "replace all". (Disclaimer: It was **not** done by me btw)

The code for `void SX1509::blink` & `void SX1509::setupBlink` seems to be different. In `void SX1509::setupBlink`, remarks are specifically mentioning that tOn for this method should not be more than 5 bits.

There seems to be some other problem.