sparkfun/SparkFun_SX1509_Arduino_Library #32: There are a couple of ooooold pull requests for this library.   How can these get merged into the master?

PKeller34, 2025-04-10:
I use the SX1509 Qwiic adapter and found a compile time issue in the .cpp a couple years ago.   I posted it here, hacked the trivial fix in my local copy way back then and promptly forgot about it.   Just now I was compiling on a new machine with the latest library and PRESTO! the same error.  Someone fixed it and issued a Pull request a couple years ago but it remains un-merged.

There is one other pull request outstanding with unrelated changes, github says all are able to be merged cleanly.

What would need to be done to get this merged in?   Surely this is a question based on ignorance, yet I *am* ignorant.

PKeller34, 2025-04-27:
Could PR26 be merged without PR 27?   I just had to reinstall the Arduino IDE, which undid my local hack to implement PR26.

Best regards,

Perry

PaulZC, 2025-04-28:
Hi Perry,

Apologies for the delay...

Resolved in v3.0.6

Best wishes,
Paul

closed 2025-04-28 by PaulZC